sogno-platform/dpsim #6: Relicense DPsim

m-mirz, 2021-01-22:
DPsim started as a GPL3 project, partly due to its dependency on VILLASnode, which is licensed under the GPL3 as well. Since the GPL3 is a strong barrier for many potential contributors, I would like to relicense the whole project under the MPL2 license.

The MPL2 is not as viral as the GPL3 and also Eigen3, one of the core libraries used in DPsim is licensed under the MPL2. Of course, I would move the code associated to VILLASnode to an external repository.

DanielDoehring, 2021-01-22:
I approve this motion to relicense DPsim under the MPL2.

junjie-zh, 2021-01-23:
I approve this motion to relicense DPsim under the MPL2.

n-eiling, 2021-01-25:
I approve this motion to relicense DPsim under the MPL2.

fwege, 2021-01-25:
I approve this motion to relicense DPsim under the MPL2

dinkelbachjan, 2021-01-25:
I approve this motion to relicense DPsim under the MPL2

hlubjuhn, 2021-01-25:
I approve this motion to relicense DPsim under the MPL2.

guelfey, 2021-01-27:
I approve this motion to relicense DPsim under the MPL2.

VivianeSaGu, 2021-01-27:
I approve this motion to relicense DPsim under the MPL2.

m-mirz, 2021-02-09:
Thank you for approving this. Then, I close this issue :-)

closed 2021-02-09 by m-mirz