sogno-platform/dpsim #23: Measure notebook runtimes

JTS22, 2021-08-02:
Some of the Jupyter-Notebooks using the new Python Interface (see #22 ) take multiple minutes for a full run-through, which causes the test pipeline to run quite long. This table aims to summarize the current run times (using the notebooks in #22 ) for evaluation and adaption of simulation run times and pipeline tests. The full notebook runtime is measured in a single run of `pytest --durations=0 -v` , the simulation run times are measured using the iPython `%%timeit` cell magic with the time calculated as the average over seven runs. Simulation run times contain the full setup and run of the dynamic simulation without PowerFlow or later data evaluation / plotting. Please note that this is not meant as a complete, accurate benchmark but rather as an overview over which notebooks should be tweaked when tested in the CI-pipeline.

## Circuit-Examples

|Notebook|Pytest runtime|Simulation runtime(s)|Asserts|Remarks|
|---|---|---|---|---|
|Compare_EMT_DP_Slack_PiLine_PQLoad|4.14s|1.73s (EMT), 419 ms (DP)|DP vs EMT||
|Compare_EMT_SynGenDQ7odTrapez_ DP_SynGenTrStab_SMIB_Fault|Skipped|48.6s (EMT), 4.25s (DP)|None|Notebook contains no error, so test could be enabled|
|CS_R2CL print_attributes|2.79s|12.5 µs (EMT), 12.5 µs (DP)|None|Notebook exists to show off attribute print functions, so comparisons with SL got stripped|
|CS_R2CL|3.36s|11.8 µs (EMT), 11.4 µs (DP)|EMT vs SL, DP vs SL||
|DP_Slack_PiLine_PQLoad_with_PF_Init|2.49s|71.6 µs|None|Plots currently plot the wrong time interval|
|DP_Slack_PiLine_VSI_Ramp|26.20s|-|None|Notebook does not use dpsimpy, plot labels are in german|
|DP_Slack_PiLine_VSI_with_PF_Init|**47.27s**|45.8s|None|wrong time interval plotted, plot labels in german|
|DP_SP_SynGenTrStab_3Bus_Fault|**32.01s**|8.94s (DP), 8.52s (SP)|None||
|DP_SP_SynGenTrStab_3Bus_SteadyState|12.71s|2.28s (DP), 2.11s (SP)|None||
|DP_SP_SynGenTrStab_SMIB_Fault|17.57s|4.06s (DP), 4.11s (SP)|None||
|DP_SP_SynGenTrStab_SMIB_SteadyState|12.29s|1.70s (DP), 1.45s (SP)|None||
|EMT_DP_Slack_PiLine_VSI_  Control_OnOff|**34.25s**|10.4s (EMT, control on), 11.2s (EMT, control off), 5.04s (DP, control on), 4.71s (DP, control off)|None||
|EMT_DP_SP_Slack_PiLine_VSI|**107.11s**|11.1s (EMT), 48s (DP), 47.6s (SP)|None||
|EMT_DP_VS_Init|3.12s|49.2ms (DP, set params), 88ms (EMT, set params), 45.1ms (DP, set attribute), 74.6ms (EMT, set attribute), 40.7ms (DP, set from node), 83.8ms (EMT, set from node)|EMT vs DP|DP vs EMT assert on set attribute commented out|
|EMT_DP_VS_RLC|19.35s|1.62s (EMT, VS+R), 518ms (DP, VS+R), 480ms (SP, VS+R), 2.82s (EMT, VS+RL), 785ms (DP, VS+RL), 856ms (SP, VS+RL), 2.79s (EMT, VS+RC), 758ms (DP, VS+RC), 865ms (SP, VS+RC)|EMT vs DP vs SP||
|EMT_Slack_PiLine_PQLoad_with_PF_Init|2.77s|493ms|None|Plots the wrong time interval|
|EMT_Slack_PiLine_VSI_with_PF_Init_Params|**69.97s**|27.9s (Kp variation), 43.7s (Ki variation), 4.33s (final choice)|None||
|EMT_Slack_PiLine_VSI_with_PF_Init|12.65s|11.3s|None||
|EMT_SynGenDQ7odTrapez_ OperationalParams_SMIB_Fault_JsonSyngenParams|**93.01s**|51.5s (no JSON-config), 50.4s (with JSON-config)|JSON-config vs No JSON-config||
|EMT_SynGenDQ7odTrapez_ OperationalParams_SMIB_Fault_ParamsModification|**94.22s**|50.8s (first parameter), 52.8s (second parameters)|None||
|EMT_SynGenDQ7odTrapez_ OperationalParams_SMIB_Fault|**93.72s**|52.9s (fundamental params), 54.5s (operational params)|Fundamental vs operational params||
|EMT_SynGenDQ7odTrapez_SMIB_Fault|**50.24s**|53.6s|None||
|RL_SourceStep_ParameterStudy|Skipped|-|None|Notebook does not use dpsimpy|
|SP_Slack_PiLine_VSI_with_PF_Init|3.69s|2.17s|None||
|VS_CS_R4|3.39s|29.8ms (EMT), 52.4ms (DP)|EMT vs SL, DP vs SL||
|VS_R2L3|3.47s|33.7ms (EMT), 63.1ms (DP)|EMT vs SL, DP vs SL||
|VS_RC1|4.22s|19.9ms (EMT), 37.5ms (DP)|EMT vs SL, DP vs SL||
|VS_RL1|3.77s|21.2ms (EMT), 33.9ms (DP)|EMT vs SL, DP vs SL||

rubensander, 2021-12-16:
<html><body>
<h2 dir="auto">Grid examples</h2>

Notebook | Pytest runtime | Simulation runtime(s) | Asserts | Remarks
-- | -- | -- | -- | --
DP_Slack_PiLine_VSI_Ramp | - | - | DP vs EMT ref | Notebook does not use dpsimpy
EMT_DP_SP_Slack_PiLine_PQLoad_FrequencyRamp_CosineFM | - | - | DP vs EMT, DP vs SP | Notebook does not use dpsimpy

</body>
</html>

m-mirz, 2022-06-29:
Next step is to remove unnecessary examples (especially C++) and reduce the runtime of the remaining ones.