sogno-platform/dpsim #37: Add compile-time option to build DPsim without libstdc++fs

stv0g, 2021-09-15:
Hi in order to build DPsim on the OPAL-RT opalrtlinux target we need to find a way to build it without linking against `libstdc++fs` as this dependency is not available on this target:

```
/usr/lib/gcc/x86_64-poky-linux/8.3.0/../../../../x86_64-poky-linux/bin/ld: cannot find -lstdc++fs
```

It appears to me that the filesytem library is not a hard requirement for building DPsim.
Can we disable certain features of DPsim which require it with "#ifdefs"?

maikwoehl, 2021-09-15:
At university we had used kernel calls of a RT patched kernel for in and out. If necessary that could be a way to write to files without the charming easy to use API of stdlib.

Just an idea on this topic.

stv0g, 2021-09-16:
Hi @maikwoehl,

what do you mean by "[...] in and out."? Are you referring to `std::cout` / `std::cin`?

maikwoehl, 2021-09-16:
Hi @stv0g,

I'm specifically referring to the [Linux Filesystem Kernel API](https://www.kernel.org/doc/html/v5.0/filesystems/index.html).

With this you do not need libstdc++-fs... unfortunately I cannot prove this fast. It has been some time ago that I have written linux kernel code.

stv0g, 2021-09-17:
Ah okay, so you run DPsim in kernel mode?

maikwoehl, 2021-09-17:
No, but you can interface to a kernel module like to a kernel driver and have your own filesystem API if you need that really hard for dpsim on OPAL-RT.

I mean you can install a [message based driver](https://www.kernel.org/doc/html/v5.0/driver-api/message-based.html) and install the output to proc or using the Linux Kernel Filesystems API.

I haven't compiled or used dpsim yet.

m-mirz, 2022-02-02:
The filesystem library is used for the logs, load profile reader and CIM file reader.

- logging could be disabled
- load profiles are not necessary for all simulations and can be communicated via VILLASnode, like the results
- the CIM reader is also not strictly necessary... the grid could be described using the C++ API

So, I think that we could disable everything related to filesystem.

stv0g, 2023-01-18:
This will be fixed by #168 

I have added a standalone implementation of `std::filesystem` via CMake FetchContent

leonardocarreras, 2023-11-05:
This can be closed due to:
#168 was merged

closed 2023-11-05 by leonardocarreras