sogno-platform/dpsim #61: Improving the attribute system

m-mirz, 2021-12-16:
Attributes should be declared as member variables of the component class to avoid something like `attribute<Complex>("V_ref")->get()` during the simulation. This is not efficient and the attributes of a component class are not easily deduced from looking at the component class declaration.

Each attribute should include its name, unit and if it is a parameter set by the user or a simulation variable. Accordingly, the python print should include more columns for this information.
```
Attribute list for object with name cs:
name                          type                size           value                         
-----------------------------------------------------------------------------------------------
I_ref                         Complex                            10.000000<0.000000            
f_src                         Real                               50.000000                     
i_intf                        MatrixReal          1x1            [...]                         
name                          String                             "cs"                          
right_vector                  MatrixReal          0x0            [...]                         
uid                           String                             "cs"                          
v_intf                        MatrixReal          1x1            [...] 
```

To make the work with the attributes more comfortable in the component classes, some operators like cast () and assignment operator might be overwritten but this should be limited as much as possible.

JTS22, 2021-12-20:
https://github.com/sogno-platform/dpsim/blob/ad491cc59c57f96738a67c882292226f579b2070/models/Include/cps/Attribute.h#L256-L267

For the new attribute system, we might also want to look at how to implement these kinds of derived attributes in a different way. Currently, these create new attributes that reference the old attribute and compute their own values using Getter and Setter functions. In this example, this behavior is inconsistent as the generated setter is never used (instead `nullptr` is passed into the constructor, crashing the program if the setter is ever called). Also, when statically looking at the code, it appears as if the Getter function calls itself (via `this->getByValue`) which would lead to an infinite loop. This code only works because `this` is captured in the lambda clause at a point where the Getter (and the getter flag) are not set yet. While technically correct, this is rather confusing and hard to understand when reading the code.

m-mirz, 2021-12-26:
The main application of the getters is to process data before sending it to VILLASweb, for example. Often we send amplitude and phase to VILLASweb instead of real and imaginary value of a complex voltage. Setters were introduced to preprocess incoming data but eventually we never used them and just made sure that the incoming data is already compliant.

Since we built the attribute system, VILLASnode introduced hooks to transform relayed data. This covers many use cases already. For logging purposes, the getters are still convenient. I could imagine that having a clean solution for the getters would be sufficient since we can use VILLASnode for other use cases.

JTS22, 2021-12-29:
All commits related to this issue will now be collected in PR #62. In the coming days I will post an overview on the (highly experimental) changes currently contained in the PR.

JTS22, 2022-01-01:
Here are some thoughts on the current state of the attribute system as well as an explanation of the new system currently included in PR #62 :
Currently, to my knowledge, there are three different kinds of attributes, which are all instances of the same `Attribute<T>` class:
- "static" attributes that own a value, which can be changed through the linked variable reference or through the `get` and `set` methods. This kind is the most common, used for example for the ID and Name of any identified object.
- "derived" attributes that do not own a value, but compute it from the value of another attribute using Getter and Setter functions. These are created by the `real` and `imag` functions on complex attributes or by the `coeff` function for matrix attributes. A derived attribute can have the same or a different type compared to the attribute it's derived from.
- "referencing" attributes. Currently these are not real attributes at all, but they are implemented by cloning the reference attribute and linking it to the attribute list of another object. Their purpose is to allow for one attribute to always have the same value as another attribute.

Additionally, dependencies can be defined for all of these attributes using the `Scheduler` and `Task` system. The various tasks describe which attributes need to be recalculated first when an attribute should be updated during the simulation.

The system introduced in PR #62 aims to unify these different types of attributes and dependencies, while allowing for full generality with regard to which attributes can be combined in which relationships. To do this, the new system provides two kinds of attributes which now have their own classes:

- `AttributeStatic<T>`: This class describes an attribute which will always own its data value and can only be changed through direct `get` and `set` functions. A static attribute can never depend on any other attribute, it can only be used as a dependency.
- `AttributeDynamic<T>`: While this attribute also contains a shared pointer to some data, this data should more be though of as some sort of value cache that might not accurately reflect the "true" value of the attribute. To compute this true value, a dynamic attribute can hold two lists of `AttributeUpdateTask`s. These tasks are invoked when either the `get` or `set` method of the dynamic attribute is called. Every task contains a list of dependency attributes as well as an actor function. This actor function receives the pointer to the dependent attribute's value cache as well as pointers to all the dependency attributes. Every actor function can therefore update the value cache (both its value as well as its pointer address) based on the values of the dependency attributes (which can be obtained via their respective `get`-methods.  An actor function can also act in reverse, reading the cached value and updating the dependency attribute's via their `set`-methods. Currently, a mix of both actions would also be possible.
When the `get` function is invoked on a dynamic attribute, it first calls all the actor functions registered for the `get` action. After these Tasks have potentially modified the cached data, a reference to this data is returned.
For the `set` function, the cached value is immediately updated to the passed value. After this, all actor functions registered for the `set` action are called, which gives them the chance to propagate the update to any dependency attributes.

Using the dynamic attributes, the "derived" and "referencing" attributes from the old model can be recreated in a more general form:
"Derived" attributes can be represented by a dynamic attribute with one task each for the `get` and `set` action. The task for `get` computes the derived value from the parent attribute(s) and updates the cache, while the task for `set` can update the parent attribute(s) based on the new value of the derived attribute. For the `real` part of a complex attribute, these tasks can look something like this:

https://github.com/JTS22/dpsim/blob/98d9b471da7665144bace970bf8ba812fb23c03c/models/Include/cps/Attribute.h#L223-L230

"Referencing" attributes are implemented using a dynamic attribute with a task which updates not the cached data value, but the cache pointer address. When referencing a static attribute, it would suffice to update the cache pointer once at the beginning of the simulation. When referencing a dynamic attribute, this update has to be performed on every `get` operation as the data pointer of the referenced attribute could also be subject to change.

To allow for this scope of generality, the resulting attribute code gets quite complicated and template-heavy. To prevent the user from having to deal with these generic types, multiple `derive` functions are introduced that create new dynamic attributes from other (static or dynamic) attributes. Using C++ 20 `concept`s, these functions can be bound to requirements such that they are only available for the attributes with the right type. This means that, for example, the `deriveReal` and `deriveImag` functions are only available for attributes with complex type. For referencing attributes, a `derive` function might be introduced that correctly installs the `get` action for copying the data pointer, either once for static references, or on every `get` for dynamic references.

Lastly, I want to stress that most of these concepts currently just exist on paper and are therefore subject to change. For example, the high amount of templates might lead to an unacceptable increase in compiled code size. Or the additional function calls for management of the actor functions might cause too much of a performance overhead to actually be useful. The system might also be too general for the use case of DPsim, in which the special cases this new system allows for might never be needed.

JTS22, 2022-02-01:
Since reworking the attributes entails reading through a lot of DPsim's current code, here are some random things I noticed that might be turned into new Issues later on:
- The MNASolver uses different attributes depending on if it computes multiple frequencies in parallel (expressed through `mFrequencyParallel)`- The corresponding attributes `mLeftSideVectorHarm` and `mLeftSideVector` should probably be unified
- Many of the basic components (Capacitor, Resistor, Inductance...) are (partly) derived from a base class which contains only the components main parameter (C, R, L...). Since these base classes are currently not derived from `AttributeList`, their attributes can not be initialized in the usual way, rather the initialization has to be moved into the subclasses. 
- The `clone` and `setParameters` methods are very similar for many basic components. Maybe a `clone` could just clone all attributes that have a certain "parameter" flag set
- To be continued...

JTS22, 2022-02-04:
Most points mentioned above are now marked in the code for PR #62 using the keywords `DEPRECATED`, `FIXME` or `CHECK`.

m-mirz, 2022-06-29:
Closed by #62. Follow up on code quality issues in new issue.

closed 2022-06-29 by m-mirz