sogno-platform/dpsim #88: Guideline on logging level choice

dinkelbachjan, 2022-03-30:
`off` - recommend to use e.g. when running case studies where exclusively the simulation results should be logged
`info` - logging of information that are basic and that are logged only once before/after the simulation
`debug` - ...
`trace` - logging of information in each simulation step

m-mirz, 2022-03-30:
This can go here: https://dpsim.fein-aachen.org/docs/development/guidelines/

dinkelbachjan, 2023-01-13:
Where should we place the logging of `LeftVector` and `RightVector`? So far, they are logged for a logging level higher than `off`
https://github.com/sogno-platform/dpsim/blob/fc169e5e41401709667ca9f2f945e906a2b9a729/dpsim/src/MNASolver.cpp#L581-L582

dinkelbachjan, 2023-01-13:
**Initial idea for how it should be:**
`off` - `scheduler.log`, `MnaSolverFactory.log`, `<SimName>_Solver.log`, `<SimName>.log`
`info` - `<Components>.log`, `<SimName>_CIM.log`
`debug` - extended `<SimName>_Solver.log` (e.g. with matrices), extended  `<SimName>_CIM.log` (e.g. with connections)
`trace` - `<SimName>_LeftVector.csv`, `<SimName>_RightVector.csv`

Note: 
- `<SimName>.csv` is a separate data logger, not influenced by `logLevel` of `Simulation` class. 
- CLI output should be independent of `logLevel`