sogno-platform/dpsim #98: Restructure and test villas examples

m-mirz, 2022-06-29:
All villas related examples should be located in examples/villas and included in the tests if possible.

JTS22, 2022-07-12:
Overview over which Villas examples actually work (tested on the `reorganization`-Branch (PR #94)):

- `dpsim-file.py`: Works without any interaction
- `dpsim-mqtt.py`: Works, but explicitly waits for input data from MQTT
- `dpsim-mqtt-import-export.py`: Works, but does not wait for input data from MQTT
- `dpsim-mqtt-import-export-MIMO.py`: Segfaults after about 3 seconds
- `shmem-distributed-reference.py`: Works (but also does not use any interfaces, as intended)
- `shmem-distributed-villas.py`: Runs til the end, but produces an error when stopping villas. All log files remain empty, so no correct results are written:
  ```
  Opening shared memory interface failed (ret=-8): No such file or directory
  villas-node: /tmp/villasnode/lib/super_node.cpp:436: villas::node::SuperNode::~SuperNode(): Assertion `state ==     State::INITIALIZED || state == State::PARSED || state == State::CHECKED || state == State::STOPPED || state ==   State::PREPARED' failed.
  Traceback (most recent call last):
  File "/dpsim/examples/villas/shmem-distributed-villas.py", line 170, in <module>
    p_villas.join()
  ```
- `test_shmem_cigre_mv_pf_profiles.py`: Starts up VillasNode and then immediately segfaults
- `test_shmem_distributed_direct.py`: Uses the old dpsim-Python interface and therefore does not run at all
- `test_shmem_import_export.py`: Starts up and segfaults after about a second
- `ShmemExample.ipynb`: The first cell runs but finishes with an error, the following cell cannot find the required log files:
  ```
  12:54:52 info node             This is VILLASnode v0.11.0-b94746e-release (built on Jul  5 2022, 15:08:48)
  12:54:52 info signals          Initialize subsystem
  12:54:52 info config           Reading configuration from local file: /tmp/tmp.pCdHd59j7J
  12:54:52 info memory           Initialize memory sub-system: #hugepages=100
  12:54:52 warn memory:mmap      Running in an unprivileged environment. Hugepages are not used!
  12:54:52 warn memory           Running in an unprivileged environment. Memory is not locked to RAM!
  12:54:52 info kernel:rt        Initialize sub-system
  12:54:52 warn kernel:rt        We recommend to use an PREEMPT_RT patched kernel!
  12:54:52 warn kernel:rt        You might want to use the 'priority' setting to increase 's process priority
  12:54:52 warn kernel:rt        You might want to use the 'affinity' setting to pin  to dedicate CPU cores
  12:54:52 info node:shmem       Initialized node type which is used by 1 nodes
  12:54:52 info node:file        Initialized node type which is used by 2 nodes
  12:54:52 info api              Starting sub-system
  12:54:52 info web              Starting sub-system
  12:54:52 info api              Started worker
  12:54:52  err lws              ERROR on binding fd 14 to port 80 (-1 98)
  12:54:52  err lws              init server failed
  12:54:52 warn web              Failed to setup vhost. Trying another port: 81
  12:54:52  err lws              ERROR on binding fd 14 to port 81 (-1 98)
  12:54:52  err lws              init server failed
  12:54:52 warn web              Failed to setup vhost. Trying another port: 82
  12:54:52 info web              Started worker
  Error while terminating subprocess (pid=83557):
  ```
- `ShmemDistributedDirect.ipynb`: Relative paths are wrong so the first cell does not run
- `cigre-mv-pf-profiles-shmem.ipynb`: Simulation setup works, but the kernel crashes on `sim.run()`

In conclusion: 4/9 Python and 0/3 Jupyter Examples actually work the way they are supposed to (and one of the working ones does not even use the Villas interface).

leonardocarreras, 2023-11-05:
I think we might want to close due to:
- PR #94 was merged

closed 2023-11-05 by leonardocarreras