sogno-platform/dpsim #109: No module named 'dpsimpy'

frostyduck, 2022-07-25:
I tried to run one of examples, but I have got the following error:

[    import dpsimpy
ModuleNotFoundError: No module named 'dpsimpy''](url)

What means dpsimpy?

m-mirz, 2022-08-10:
Hi @frostyduck, it looks like the dpsimpy module is not available in the PYTHONPATH.
The dpsimpy module is created using pybind and exposes all C++ core functions of DPsim to Python.

If you are using the [dev Dockerfile](https://github.com/sogno-platform/dpsim/blob/master/packaging/Docker/Dockerfile.dev), you need to add the location of the modules to the PYTHONPATH environment variable using 'export' as explained in the [build instructions](https://dpsim.fein-aachen.org/docs/getting-started/build/).

If you are using [this](https://github.com/sogno-platform/dpsim/blob/master/packaging/Docker/Dockerfile) Dockerfile, the python package is automatically installed and everything should work out of the box.

Please feel free to message me in the LFE slack in case my hints do not solve your problem: https://slack.lfenergy.org/

m-mirz, 2022-11-12:
I close this issue for now. Please come back to me if the issue persists for your installation.

closed 2022-11-12 by m-mirz