sogno-platform/dpsim #127: Automated pypi package deployment

dinkelbachjan, 2022-10-26:


dinkelbachjan, 2022-10-26:
@m-mirz We would like to consider start with the automated pypi package deployment. Was this manually done so far? Or is there any starting point already somewhere?

JTS22, 2022-10-30:
The deployment can probably be implemented using GitHub actions as described in https://packaging.python.org/en/latest/guides/publishing-package-distribution-releases-using-github-actions-ci-cd-workflows/ . The `python -m  build --sdist --wheel --outdir dist/` command already works for dpsim. Given that the generated package has the correct format for PyPi, the upload should be quite easy to implement.

dinkelbachjan, 2022-11-02:
@m-mirz Procedure for new DPsim version:
    - Create a new tag with increased version number
    - Update setup.cfg
    - Update CMakeLists.txt
    - Update Dockerfile
A new pypi package will be then deployed accordingly.

PR #139 (2022-11-28): Add Github actions workflow for publishing to PyPi
Adds a new workflow for packaging the dpsim source distribution as well as building multiple binary wheels for Python versions 3.6 to 3.11 using [cibuildwheel](https://github.com/pypa/cibuildwheel). Every commit to the master branch will be pushed to TestPyPi while only tagged commits will also be uploaded to production PyPi.

Closes #127
closed 2022-11-28 by dinkelbachjan