sogno-platform/dpsim #130: System Matrix Stamp of DP_Ph3_VoltageSource

cwirtz-fgh, 2022-11-09:
### **System Matrix Stamp of DP_Ph3_VoltageSource**
In DP_Ph3_VoltageSource Line 69 the stamp for phase C differs from A and B:

`if (terminalNotGrounded(0)) {
...

		Math::addToMatrixElement(systemMatrix, matrixNodeIndex(0, 1), mVirtualNodes[0]->matrixNodeIndex(PhaseType::B), Complex(-1, 0));
		Math::addToMatrixElement(systemMatrix, mVirtualNodes[0]->matrixNodeIndex(PhaseType::B), matrixNodeIndex(0, 1), Complex(-1, 0));

		Math::addToMatrixElement(systemMatrix, matrixNodeIndex(0, 2), mVirtualNodes[0]->matrixNodeIndex(PhaseType::C), Complex(-1, 0));
		Math::addToMatrixElement(systemMatrix, mVirtualNodes[0]->matrixNodeIndex(PhaseType::C), matrixNodeIndex(0, 2), -Complex(-1, 0));
	}`

This leads to phase C beeing shifted when transformed into EMT the frame and unintended model behaviour:
Bug ( with -Complex(-1,0)); )
[image]

Fixed (with Complex(-1,0)); )
[image]

m-mirz, 2022-11-12:
Fixed in #137
To be reviewed by @dinkelbachjan

m-mirz, 2022-11-15:
Closed by #137

closed 2022-11-15 by m-mirz