sogno-platform/dpsim #131: Every SimPowerComp has a vector mSubComponents of sub-components but most classes derived from SimPowerComp have their sub-components as separate member-variables, not utilizing this list.

dinkelbachjan, 2022-11-09:


JTS22, 2022-11-17:
Classes which include subcomponents:
| Class | Uses `mSubComponents` | Pre- and Post-step called by parent? |
|--------------|-----------|------------|
|DP_Ph1_VoltageSourceInverterDQ|yes|yes|
|DP_Ph1_NetworkInjection|yes|yes|
|DP_Ph1_PiLine|no|yes|
|DP_Ph1_PQLoadCS|no|no|
|DP_Ph1_RxLine|no|no|
||||

JTS22, 2022-11-25:
In addition to the changes made in #141, I propose another way to further unify the behavior of components with subcomponents: By creating a shared superclass MNASimPowerComp that is derived from both SimPowerComp and MNAInterface, it becomes possible to extract the various MNA method calls on the subcomponents into this common superclass. This means that the parent components only register their subcomponents once (e. g. through `addMNASubComponent`) and then only contain methods concerned with the MNA operations in the parent. The calls to the MNA methods of the child are handled by the superclass instead, drastically cutting down on the amount of duplicated code (especially the `for (auto comp : mSubComponents)` loop) in the parent classes. The `MNAPreStep` and `MNAPostStep` classes contained in almost all parent components can also be merged into the common superclass. A first working example of this can be found in https://github.com/sogno-platform/dpsim/tree/unify-mna-subcomponents (Comparison with #141 in https://github.com/sogno-platform/dpsim/compare/subcomponents...unify-mna-subcomponents).

dinkelbachjan, 2022-11-28:
@m-mirz We are thinking of inserting a new class `CompositePowerComp` as described above to unify the subcomponent handling. What do you think about that?

PR #159 (2023-02-28): Update and expand documentation
Documentation for subcomponents describes the behavior added in #141 
Documentation for mna components describes the behavior added in #150 

Closes #133 
Closes #135 
Closes #131
commit da1dfad7 (2023-02-28): Update and expand documentation (#159)

Documentation for subcomponents describes the behavior added in #141 
Documentation for mna components describes the behavior added in #150 

Closes #133 
Closes #135 
Closes #131
closed 2023-02-28 by m-mirz