sogno-platform/dpsim #132: Go through and update all FIXME, CHECK, TODO, DEPRECATED and THISISBAD comments in the code

dinkelbachjan, 2022-11-09:


JTS22, 2022-11-27:
I started working on FIXMEs in #142

JTS22, 2022-11-28:
https://github.com/sogno-platform/dpsim/blob/0a9d00cefcc75ccd51072e41619ed2e755e265f5/dpsim/src/MNASolver.cpp#L40-L52
This is probably harder to solve because one needs to rework the MNA Solver for parallel frequencies (if that is even a working usecase right now)

JTS22, 2022-11-28:
https://github.com/sogno-platform/dpsim/blob/0a9d00cefcc75ccd51072e41619ed2e755e265f5/dpsim-models/include/dpsim-models/Signal/DecouplingLine.h#L49-L50
https://github.com/sogno-platform/dpsim/blob/0a9d00cefcc75ccd51072e41619ed2e755e265f5/dpsim-models/include/dpsim-models/Signal/DecouplingLineEMT.h#L49-L50

I dont know what states this is talking about. The Matrix attribute itself is only used as a task dependency but never read or written to...

JTS22, 2022-11-28:
https://github.com/sogno-platform/dpsim/blob/0a9d00cefcc75ccd51072e41619ed2e755e265f5/dpsim-models/src/DP/DP_Ph1_ResIndSeries.cpp#L16-L17
Removing the R attribute seems to go against the purpose of the class. But then again, the class is never used / not even compiled so maybe we can just delete it altogether?

JTS22, 2022-11-28:
https://github.com/sogno-platform/dpsim/blob/0a9d00cefcc75ccd51072e41619ed2e755e265f5/examples/Notebooks/Components/Trafo.ipynb?short_path=8ca3d82#L69-L71
I cannot comment on this, should be easy to resolve though.

JTS22, 2022-11-28:
https://github.com/sogno-platform/dpsim/blob/0a9d00cefcc75ccd51072e41619ed2e755e265f5/examples/Notebooks/Components/SynGenDq7od_ParameterStudy.ipynb?short_path=5253b86#L454
The notebook does not even run the simulation to completion, so this seems like a secondary issue...

JTS22, 2022-11-28:
https://github.com/sogno-platform/dpsim/blob/0a9d00cefcc75ccd51072e41619ed2e755e265f5/examples/Notebooks/Components/SynGen_trStab.ipynb?short_path=7c394d7#L35
I can confirm this is still an issue, there is even an entire notebook for examination: https://github.com/sogno-platform/dpsim/blob/master/examples/Notebooks/Components/SynGen_trStab_logger_test.ipynb
The core issue remains unsolved though, probably something to do with the logger only writing out files when the object is deleted from memory...