sogno-platform/dpsim #133: Documentation attribute system

dinkelbachjan, 2022-11-09:


m-mirz, 2022-11-27:
Hugo does not require themes as submodules anymore. You can now include them as go modules: https://gohugo.io/hugo-modules/
Besides, Github added support for latex in markdown files. So you can directly view the equations even without a static website generator :-)

I would be open to move the entire dpsim docs into the dpsim code repo. What do you think? @dinkelbachjan @JTS22

JTS22, 2022-11-28:
I do not know exactly how the docs are currently hosted, but including them in the dpsim code repo makes sense to me. This would allow to find and read them directly on GitHub and to transparently update them with every (code) pull request.

dinkelbachjan, 2022-11-28:
Yes, I would also prefer that, hoping that this also reduces the barrier to contribute to them

PR #159 (2023-02-28): Update and expand documentation
Documentation for subcomponents describes the behavior added in #141 
Documentation for mna components describes the behavior added in #150 

Closes #133 
Closes #135 
Closes #131
commit da1dfad7 (2023-02-28): Update and expand documentation (#159)

Documentation for subcomponents describes the behavior added in #141 
Documentation for mna components describes the behavior added in #150 

Closes #133 
Closes #135 
Closes #131
closed 2023-02-28 by m-mirz