sogno-platform/dpsim #145: Update  VILLASnode to be able to compile with GCC 12

JTS22, 2022-11-28:
The version currently used in the DPsim docker images (Villas commit [b94746effb015aa98791c0e319ef11223d18e8b0](https://github.com/VILLASframework/VILLASnode/tree/b94746effb015aa98791c0e319ef11223d18e8b0)) uses VILLAScommon on commit 25cd53ee6882c3f66746d6d8c27790ef22d18322 as a submodule. This version cannot be compiled with GCC version 12+. Apparently, the repos used in the new ManyLinux Docker image (added in #139 ) changed during the last days to now download GCC 12 instead of 11. This broke the Docker image build between https://github.com/sogno-platform/dpsim/actions/runs/3480565146/jobs/5821064882 and https://github.com/sogno-platform/dpsim/actions/runs/3565405165/jobs/5990523552. Because the current master commit of VILLASnode also still uses the same subcommit of VILLAScommon, just updating the version of VILLASnode in dpsim will not suffice, but either dpsim needs to manually checkout a newer subcommit of VILLAScommon (probably not a good idea) or the VILLASnode master needs to include a newer subcommit.

stv0g, 2023-01-10:
@JTS22 Could you file an issue with [VILLAScommon ](https://github.com/VILLASframework/common/)to figure out why it does not build on GCC 12?

JTS22, 2023-07-05:
The compilation with GCC12 seems to be fixed on the newest master commit b511424. The CI still seems to use GCC11 though (see https://github.com/sogno-platform/dpsim/actions/runs/5454599508/jobs/9924938203). Compilation with clang (v. 14.0.5) still does not work though, even when not treating warnings as errors.

stv0g, 2023-07-05:
I think we should have fixed the issue for VILLASnode though.

JTS22, 2023-07-05:
Yes, this is now purely a dpsim problem :) Guess I will make a new issue for it.

closed 2023-07-05 by JTS22