sogno-platform/dpsim #156: Improve documentation UX

JTS22, 2022-12-11:
- [ ] Although Hugo seems to support syntax highlighting via `pygments`, the code contained in the documentation is currently not correctly highlighted
- [ ] The actual markdown files are contained in many nested folders which do not directly correspond to the generated HTML page tree. This makes it hard to track changes, especially when most of the files are named `index.html`
- [ ] The links between pages only work in the built HTML version, but cannot be used to jump between markdown files on GitHub.

Many of these issues could be solved by using other tools better suited for building code documentation, e. g. [mdbook](https://rust-lang.github.io/mdBook/index.html). This results in a file structure similar to here: https://github.com/rust-lang/nomicon/tree/master/src

m-mirz, 2022-12-21:
I am fixing the equations for katex and adding missing images in https://github.com/sogno-platform/dpsim/pull/165

m-mirz, 2022-12-22:
@dinkelbachjan This feature could be interesting for the discussion about a new solver class structure: https://www.docsy.dev/docs/adding-content/diagrams-and-formulae/#uml-diagrams-with-plantuml