sogno-platform/dpsim #173: Make parallelism is to high by default

stv0g, 2023-01-11:
DPsim's build system uses `make -j$(nproc)` in many places which undesirable as this degree of parallelism is quite high for some systems.

I see that my machine (64GB, 16-core i7 workstation) quickly fills up all of the memory and slows down.
So a `make -j6` is a lot faster than a `make -j16` as we dont need to do demand paging..

I propose the make the make parallelism configurable via a environment variable named `MAKE_PROCS`

dinkelbachjan, 2023-01-24:
Thanks, that's very good to know! I agree with your proprosal. Should we go for a default value of 8 then?

stv0g, 2023-01-24:
I think 8 could be a reasonable default. But its also more important that we can easily change this in the scripts by settings an environment variable. E.g. `MAKE_PROCS` as mentioned above.

n-eiling, 2023-01-27:
Cmake also has -j right?
```
-j [<jobs>], --parallel [<jobs>]

    New in version 3.12.

    The maximum number of concurrent processes to use when building. If <jobs> is omitted the native build tool's default number is used.

    The [CMAKE_BUILD_PARALLEL_LEVEL](https://cmake.org/cmake/help/latest/envvar/CMAKE_BUILD_PARALLEL_LEVEL.html#envvar:CMAKE_BUILD_PARALLEL_LEVEL) environment variable, if set, specifies a default parallel level when this option is not given.

    Some native build tools always build in parallel. The use of <jobs> value of 1 can be used to limit to a single job.
    ```

stv0g, 2023-01-30:
Yes, we might want to switch to `cmake --build` anyways instead of calling `make` as there might be platforms for which make is not the default CMake build tool