sogno-platform/dpsim #180: Scheduler keep dropping the task

BeijiaYang, 2023-01-26:
**Describe the bug**
I have create a new signal model: "PMUSignalDevice", and I try to apply it on a simple simulation "three bus system". The new class has only one function, which is add to error/noise on the input. Normally the input is signal from a system component.
The problem is, however, the scheduler keeps dropping the task of the new class.

**To Reproduce**
Steps to reproduce the behavior:
To see the PMUSignalDevice.h:
**https://github.com/BeijiaYang/dpsim/blob/pmu-signal/dpsim-models/include/dpsim-models/Signal/PMUSignalDevice.h**
To see the Three bus system:
**https://github.com/BeijiaYang/dpsim/blob/pmu-signal/dpsim/examples/cxx/Circuits/Three_bus_sim.cpp**

Problem solved:
With the help of an senior engineer, the problem is solved. The output of the class is defined as a Attribute(matrixcomp). The initalisation should be Zero(1,1).

cwirtz-fgh, 2023-01-31:
Regarding the task dependencies it would be helpful to extend the documentation and explain in detail the system of task dependencies, which ones are dropped and why. 
https://dpsim.fein-aachen.org/docs/tasks/add-model/

As a workaround (because i didnt understand why tasks were dropped) I used the "external" attribute in several models:
```
static CPS::AttributeBase::Ptr external;
modifiedAttributes.push_back(external);
```

dinkelbachjan, 2023-02-09:
Yes, I fully agree. Let's follow up on this in #188. I guess @m-mirz could have some material suitable to extend the docu on the task system.

dinkelbachjan, 2023-02-09:
The issue is that the logger drops the attribute if it cannot derive its dimensions. The dimensions could not be derived, because the attribute has not been initialised. Most importantly, this behavior should not happen silently.