sogno-platform/dpsim #219: Broken literature references in documentation

stv0g, 2023-05-15:
https://sogno.energy/dpsim/docs/concepts/dyn-phasors/#bandpass-signals-and-baseband-representation

georgii-tishenin, 2024-06-20:
Hi, @stv0g. Are you talking about the references [Maas2003, Suarez2009, Haykin2009, Proakis2001]? I can add the links.

stv0g, 2024-06-20:
@georgii-tishenin Yes, those are the ones :)

georgii-tishenin, 2024-06-22:
I found [Maas2003](https://ieeexplore.ieee.org/document/9100822) and [Suarez2009](https://ieeexplore.ieee.org/book/5236646) on IEEE Xplore. 

For [Haykin2009](https://www.wiley.com/en-us/Communication+Systems%2C+5th+Edition-p-9780471697909) and [Proakis2001](https://www.abebooks.com/9780131589322/Communication-Systems-Engineering-Proakis-John-0131589326/plp) I could only find links to book stores. For those two could be better to just provide full references at the end of the page:
### References without links

- [Haykin2009] S. Haykin, Communication systems. John Wiley & Sons, 2008.

- [Proakis2001] J. G. Proakis, M. Salehi, N. Zhou, et al., Communication systems engineering. Prentice Hall New Jersey, 1994, vol. 2

What do you say? @stv0g, @m-mirz

stv0g, 2024-06-22:
Ideally I would link to doi.org

This always redidirects to the publisher.

m-mirz, 2024-06-22:
I agree with @stv0g

georgii-tishenin, 2024-06-23:
It looks like doi is not a must for books. And, unfortunately, neither of Maas2003, Suarez2009, Haykin2009, Proakis2001 have doi😞.
@m-mirz, @stv0g let me know what you'd prefer after doi (and sorry for a long discussion over this simple topic).
Personally, for [Maas2003](https://ieeexplore.ieee.org/document/9100822) and [Suarez2009](https://ieeexplore.ieee.org/book/5236646) I'd link to IEEE Xplore. As Haykin2009, Proakis2001 are not on IEEE Xplore either, one more possible idea for them is to link to [https://isbnsearch.org/](https://isbnsearch.org/), as they have ISBN.

m-mirz, 2024-06-23:
ISBN is a good alternative I think @georgii-tishenin

PR #310 (2024-06-27): Fix docu issues
This PR fixes two small documentation issues:

- Substitute duplicate `ODESolver` by `DAESolver` in UML diagram [dpsim_classes_simulation.svg](https://github.com/sogno-platform/dpsim/blob/master/docs/hugo/content/en/docs/Overview/Architecture/dpsim_classes_simulation.svg), used in [Architecture | DPsim](https://dpsim.fein-aachen.org/docs/overview/architecture/). Fixes #308 .
- Add missing links for literature references in [Dynamic Phasors | DPsim](https://sogno.energy/dpsim/docs/concepts/dyn-phasors/#bandpass-signals-and-baseband-representation). Fixes #219 .
commit 204eb276 (2024-06-27): Fix docu issues (#310)

This PR fixes two small documentation issues:

- Substitute duplicate `ODESolver` by `DAESolver` in UML diagram
[dpsim_classes_simulation.svg](https://github.com/sogno-platform/dpsim/blob/master/docs/hugo/content/en/docs/Overview/Architecture/dpsim_classes_simulation.svg),
used in [Architecture |
DPsim](https://dpsim.fein-aachen.org/docs/overview/architecture/). Fixes
#308 .
- Add missing links for literature references in [Dynamic Phasors |
DPsim](https://sogno.energy/dpsim/docs/concepts/dyn-phasors/#bandpass-signals-and-baseband-representation).
Fixes #219 .
closed 2024-06-27 by m-mirz