sogno-platform/dpsim #231: Villas Examples do not run in current Dev container

JTS22, 2023-07-17:
When using the dev-container from the current `Dockerfile.dev`, none of the VillasNode-Examples can be executed, because they require the presence of the `libxil` shared library (https://github.com/VILLASframework/libxil). Because the last push of the dev image to Docker hub happened over 6 months ago, the pipeline still uses an outdated image, in which this dependency was apparently not present. Therefore, the Villas-Examples still work in the GitHub-Pipeline, but not locally.
Options to resolve this are:
- Compile VillasNode with the `WITH_FPGA=OFF` option to disable the dependency
- Somehow install the dependency manually in the dpsim-Dockerfile
- Update VillasNode (which should probably happen anyway at some point). This resolves the issue, but introduces other problems, so the dpsim-code would need to be ported too.

dinkelbachjan, 2023-07-27:
We target to implement the third option.

dinkelbachjan, 2023-10-20:
The VILLASnode version and, correspondingly, the Docker hub dev image have been updated with the merge of PR #245

closed 2023-10-20 by dinkelbachjan