sogno-platform/dpsim #285: Delete stale branches

stv0g, 2024-04-24:
The DPsim repo on GitHub has currently quite a number of stale branches. We should aim at removing the number of open branches and delete or merge them wherever possible.

Here is list of open branches:

- [ ] [`255-pypi-build-and-deploy-fails`](https://github.com/sogno-platform/dpsim/compare/255-pypi-build-and-deploy-fails)
- [ ] [`ACDC_HVDCTests`](https://github.com/sogno-platform/dpsim/compare/ACDC_HVDCTests)
- [ ] [`ControlBlocks`](https://github.com/sogno-platform/dpsim/compare/ControlBlocks)
- [ ] [`DP-vsi-voltage-control`](https://github.com/sogno-platform/dpsim/compare/DP-vsi-voltage-control)
- [ ] [`DP-vsi-voltage-control-rebase`](https://github.com/sogno-platform/dpsim/compare/DP-vsi-voltage-control-rebase)
- [ ] [`IterativeMNASolverDirect`](https://github.com/sogno-platform/dpsim/compare/IterativeMNASolverDirect)
- [ ] [`Sundials`](https://github.com/sogno-platform/dpsim/compare/Sundials)
- [ ] [`ba-can`](https://github.com/sogno-platform/dpsim/compare/ba-can)
- [ ] [`benchmark-networks`](https://github.com/sogno-platform/dpsim/compare/benchmark-networks)
- [ ] [`benchmark-networks-cosim`](https://github.com/sogno-platform/dpsim/compare/benchmark-networks-cosim)
- [ ] [`centos-dockerfile`](https://github.com/sogno-platform/dpsim/compare/centos-dockerfile)
- [ ] [`ci-minimal-build`](https://github.com/sogno-platform/dpsim/compare/ci-minimal-build)
- [ ] [`compiler-warning-fixes-256`](https://github.com/sogno-platform/dpsim/compare/compiler-warning-fixes-256)
- [ ] [`cosim-examples`](https://github.com/sogno-platform/dpsim/compare/cosim-examples)
- [ ] [`cosim-examples-rebase`](https://github.com/sogno-platform/dpsim/compare/cosim-examples-rebase)
- [ ] [`cwirtz-fgh-EMT_Ph1_Switch`](https://github.com/sogno-platform/dpsim/compare/cwirtz-fgh-EMT_Ph1_Switch)
- [ ] [`dependabot/npm_and_yarn/docs/hugo/yaml-2.2.2`](https://github.com/sogno-platform/dpsim/compare/dependabot/npm_and_yarn/docs/hugo/yaml-2.2.2)
- [ ] [`deploy-docker`](https://github.com/sogno-platform/dpsim/compare/deploy-docker)
- [ ] [`deploy-docs-test`](https://github.com/sogno-platform/dpsim/compare/deploy-docs-test)
- [ ] [`dp-9-bus-dyn-switch-nb`](https://github.com/sogno-platform/dpsim/compare/dp-9-bus-dyn-switch-nb)
- [ ] [`droop-control`](https://github.com/sogno-platform/dpsim/compare/droop-control)
- [ ] [`eigenvalues`](https://github.com/sogno-platform/dpsim/compare/eigenvalues)
- [ ] [`emt-syngen-trstab`](https://github.com/sogno-platform/dpsim/compare/emt-syngen-trstab)
- [ ] [`fix-cmake`](https://github.com/sogno-platform/dpsim/compare/fix-cmake)
- [ ] [`fix-devcontainer`](https://github.com/sogno-platform/dpsim/compare/fix-devcontainer)
- [ ] [`fix-docs-leo`](https://github.com/sogno-platform/dpsim/compare/fix-docs-leo)
- [ ] [`fix-doxygen-input-paths`](https://github.com/sogno-platform/dpsim/compare/fix-doxygen-input-paths)
- [ ] [`fix-sonar-jdk`](https://github.com/sogno-platform/dpsim/compare/fix-sonar-jdk)
- [ ] [`fix_initialization_rxload`](https://github.com/sogno-platform/dpsim/compare/fix_initialization_rxload)
- [ ] [`fix_power_flow_initialisation`](https://github.com/sogno-platform/dpsim/compare/fix_power_flow_initialisation)
- [ ] [`gh-pages`](https://github.com/sogno-platform/dpsim/compare/gh-pages)
- [ ] [`grid-forming-inverter`](https://github.com/sogno-platform/dpsim/compare/grid-forming-inverter)
- [ ] [`grid-forming-inverter-rebase`](https://github.com/sogno-platform/dpsim/compare/grid-forming-inverter-rebase)
- [ ] [`hotfix-missing-simulation-stop`](https://github.com/sogno-platform/dpsim/compare/hotfix-missing-simulation-stop)
- [ ] [`ieee14-v2`](https://github.com/sogno-platform/dpsim/compare/ieee14-v2)
- [ ] [`improve-logging`](https://github.com/sogno-platform/dpsim/compare/improve-logging)
- [ ] [`inergy-demo`](https://github.com/sogno-platform/dpsim/compare/inergy-demo)
- [ ] [`iterative_syngens_optim_lsc`](https://github.com/sogno-platform/dpsim/compare/iterative_syngens_optim_lsc)
- [ ] [`jl-mna-plugin`](https://github.com/sogno-platform/dpsim/compare/jl-mna-plugin)
- [ ] [`jl-mna-plugin-32bit`](https://github.com/sogno-platform/dpsim/compare/jl-mna-plugin-32bit)
- [ ] [`klu-default`](https://github.com/sogno-platform/dpsim/compare/klu-default)
- [ ] [`klu-solver-error`](https://github.com/sogno-platform/dpsim/compare/klu-solver-error)
- [ ] [`klu-testing`](https://github.com/sogno-platform/dpsim/compare/klu-testing)
- [ ] [`klu-testing-cr`](https://github.com/sogno-platform/dpsim/compare/klu-testing-cr)
- [ ] [`klu-testing-pr`](https://github.com/sogno-platform/dpsim/compare/klu-testing-pr)
- [ ] [`kundur-two-area`](https://github.com/sogno-platform/dpsim/compare/kundur-two-area)
- [ ] [`libxil-dep`](https://github.com/sogno-platform/dpsim/compare/libxil-dep)
- [ ] [`mat2dpsimpy`](https://github.com/sogno-platform/dpsim/compare/mat2dpsimpy)
- [ ] [`mnasolver-nicslu-adapter`](https://github.com/sogno-platform/dpsim/compare/mnasolver-nicslu-adapter)
- [ ] [`mybinder`](https://github.com/sogno-platform/dpsim/compare/mybinder)
- [ ] [`no-actions-on-pr`](https://github.com/sogno-platform/dpsim/compare/no-actions-on-pr)
- [ ] [`opal-rtlab`](https://github.com/sogno-platform/dpsim/compare/opal-rtlab)
- [ ] [`pju-devcontainer`](https://github.com/sogno-platform/dpsim/compare/pju-devcontainer)
- [ ] [`pmu-signal`](https://github.com/sogno-platform/dpsim/compare/pmu-signal)
- [ ] [`powerflow-revision`](https://github.com/sogno-platform/dpsim/compare/powerflow-revision)
- [ ] [`pyproject`](https://github.com/sogno-platform/dpsim/compare/pyproject)
- [ ] [`python-logsteptimes`](https://github.com/sogno-platform/dpsim/compare/python-logsteptimes)
- [ ] [`remove-rwth-gitlab-deps`](https://github.com/sogno-platform/dpsim/compare/remove-rwth-gitlab-deps)
- [ ] [`revise-pf-solver`](https://github.com/sogno-platform/dpsim/compare/revise-pf-solver)
- [ ] [`rl-element`](https://github.com/sogno-platform/dpsim/compare/rl-element)
- [ ] [`rl-element-rebase`](https://github.com/sogno-platform/dpsim/compare/rl-element-rebase)
- [ ] [`rm-gitlab-docker-build`](https://github.com/sogno-platform/dpsim/compare/rm-gitlab-docker-build)
- [ ] [`sg-controllers`](https://github.com/sogno-platform/dpsim/compare/sg-controllers)
- [ ] [`slew-scenarios-all`](https://github.com/sogno-platform/dpsim/compare/slew-scenarios-all)
- [ ] [`solver-parameters`](https://github.com/sogno-platform/dpsim/compare/solver-parameters)
- [ ] [`ssn`](https://github.com/sogno-platform/dpsim/compare/ssn)
- [ ] [`ssn-linear`](https://github.com/sogno-platform/dpsim/compare/ssn-linear)
- [ ] [`syngen-vbr-nicslu`](https://github.com/sogno-platform/dpsim/compare/syngen-vbr-nicslu)
- [ ] [`unify-mna-subcomponents-old`](https://github.com/sogno-platform/dpsim/compare/unify-mna-subcomponents-old)
- [ ] [`update-contacts`](https://github.com/sogno-platform/dpsim/compare/update-contacts)
- [ ] [`update-repo-refs`](https://github.com/sogno-platform/dpsim/compare/update-repo-refs)
- [ ] [`update-villas-version`](https://github.com/sogno-platform/dpsim/compare/update-villas-version)
- [ ] [`villas-worker`](https://github.com/sogno-platform/dpsim/compare/villas-worker)
- [ ] [`voltage-source-delay`](https://github.com/sogno-platform/dpsim/compare/voltage-source-delay)
- [ ] [`vsi-voltage-control`](https://github.com/sogno-platform/dpsim/compare/vsi-voltage-control)
- [ ] [`vsi-voltage-control-RMS-fix`](https://github.com/sogno-platform/dpsim/compare/vsi-voltage-control-RMS-fix)

georgii-tishenin, 2024-06-07:
Hi, @stv0g, thanks for creating this issue.
To move this forward and eventually close the issue we can:
**1. Identify Stale Branches**
- We can use [this list of stale branches](https://github.com/sogno-platform/dpsim/branches/stale). These are the branches with >3 months without a commit. Or we could may be use our own more relaxed definition of stale, e.g., 1 year without a commit.

**2. Assign Ownership**
- Assign each stale branch to its creator.

 **3. Set a Response Deadline**
- Give creators 1 month to confirm if their branch is needed or can be deleted.

**4. Delete Unclaimed Branches**
- Delete branches with no response after the deadline.

what do you think?

stv0g, 2024-09-17:
There are the following stale branches created by you. Can we delete them, or are they still beeing worked on / going to be merged at some point in time?

@dinkelbachjan:

- https://github.com/sogno-platform/dpsim/tree/slew-scenarios-all
- https://github.com/sogno-platform/dpsim/tree/powerflow-revision
- https://github.com/sogno-platform/dpsim/tree/ssn

@LennartSchu 

- https://github.com/sogno-platform/dpsim/tree/syngen-vbr-nicslu
- https://github.com/sogno-platform/dpsim/tree/iterative_syngens_optim_lsc
- https://github.com/sogno-platform/dpsim/tree/mnasolver-nicslu-adapter

@m-mirz 

- https://github.com/sogno-platform/dpsim/tree/dp-9-bus-dyn-switch-nb
- https://github.com/sogno-platform/dpsim/tree/fix-devcontainer

@martinmoraga 

- https://github.com/sogno-platform/dpsim/tree/Sundials
- https://github.com/sogno-platform/dpsim/tree/ieee14-v2
- https://github.com/sogno-platform/dpsim/tree/solver-parameters
- https://github.com/sogno-platform/dpsim/tree/vsi-voltage-control-RMS-fix
- https://github.com/sogno-platform/dpsim/tree/ba-can
- https://github.com/sogno-platform/dpsim/tree/rl-element
- https://github.com/sogno-platform/dpsim/tree/grid-forming-inverter
- https://github.com/sogno-platform/dpsim/tree/rl-element-rebase
- https://github.com/sogno-platform/dpsim/tree/benchmark-networks
- https://github.com/sogno-platform/dpsim/tree/grid-forming-inverter-rebase
- https://github.com/sogno-platform/dpsim/tree/DP-vsi-voltage-control-rebase2

@georgii-tishenin 

- https://github.com/sogno-platform/dpsim/tree/log-at-debug-level

@philipp-fensch 

- https://github.com/sogno-platform/dpsim/tree/klu-testing
- https://github.com/sogno-platform/dpsim/tree/klu-solver-error
- https://github.com/sogno-platform/dpsim/tree/klu-testing-cr
- https://github.com/sogno-platform/dpsim/tree/klu-testing-cr
- https://github.com/sogno-platform/dpsim/tree/klu-testing-pr
- https://github.com/sogno-platform/dpsim/tree/python-logsteptimes

@fwege 

- https://github.com/sogno-platform/dpsim/tree/jl-mna-plugin

This is just a first bunch of the branches. I would say, if we dont receive any feedback within the next 4 weeks, we should delete the branches..

georgii-tishenin, 2024-09-19:
Hi, @stv0g, thanks for driving this further. I removed my stale branch👍

m-mirz, 2024-09-20:
@stv0g I have moved my branches to my fork as commented in the other PR. Before you delete branches, please let me know which ones. I would like to store them in my fork as well if the original contributors do not follow up on them. I know that considerable effort went into some of them and do not want to lose the work.

LennartSchu, 2024-09-20:
@stv0g I'm not working on any of those branches. I'm not sure either whether either can be deleted. At least one of them is associated with [a PR](https://github.com/sogno-platform/dpsim/pull/204l), but I don't know if it will ever be merged. The branches seem to be too divergent from the current `master`-branch to be useful, I guess.

stv0g, 2024-09-20:
@m-mirz @LennartSchu I would delete the branches mentioned in my previous comment. There are still more, but I would like to delete them in smaller batches rather than all at once.

m-mirz, 2024-09-22:
@stv0g I have stored all of the branches above in my fork as well. So, you can go ahead deleting them if you do not receive feedback.

stv0g, 2024-11-13:
The first bunch of branches [from my previous comment](https://github.com/sogno-platform/dpsim/issues/285#issuecomment-2355813836) have been removed.

I will delete the second batch in on the 2024-12-13 if they respective authors dont reach out in this thread:

@gnakti

- https://github.com/sogno-platform/dpsim/tree/kundur-two-area
- https://github.com/sogno-platform/dpsim/tree/vsi-voltage-control
- https://github.com/sogno-platform/dpsim/tree/emt-syngen-trstab

@JTS22 

- https://github.com/sogno-platform/dpsim/tree/libxil-dep
- https://github.com/sogno-platform/dpsim/tree/improve-logging

@matthiasmees

- https://github.com/sogno-platform/dpsim/tree/droop-control
- https://github.com/sogno-platform/dpsim/tree/ControlBlocks

@Shreya210799

- https://github.com/sogno-platform/dpsim/tree/ACDC_HVDCTests

@leonardocarreras 

- https://github.com/sogno-platform/dpsim/tree/pmu-signal
- https://github.com/sogno-platform/dpsim/tree/inergy-demo

stv0g, 2024-11-13:
As discussed in our last DPsim meeting, I also ask all other authors to move branches to their own forks of the DPsim repo and open PRs from there.

gnakti, 2024-11-13:
@stv0g I moved my branches and those of @matthiasmees and @Shreya210799 to my fork.

MarvinTollnitschRWTH, 2024-12-22:
The branch 'IterativeMNASolverDirect' has been recreated in my fork and can be deleted here!

m-mirz, 2025-01-24:
> The branch 'IterativeMNASolverDirect' has been recreated in my fork and can be deleted here!
> 
> The branch on my fork somehow compares with the one here though, I hope this is not an issue: [image]

This seems normal. It is the same for me.

stv0g, 2025-01-28:
Second batch of branches has been deleted. All remaining branches are less than a year old.
I think we can keep them for now.

Policy for all future work should remain: Create branches in personal forks of the repo, then create a PR into this upstream repo :)

Thanks for you support!

closed 2025-01-28 by stv0g