sogno-platform/dpsim #287: Introduce a `CODEOWNERS` file

stv0g, 2024-04-24:
See: https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-code-owners

As discussed in our meeting on the 23rd April:

We would like to introduce a `CODEOWNERS` file in the repo to more easily assign the task of reviewing to the corresponding experts of the changed code.

Lets first discuss how the initial file should look like.

I would like to ask everybody to indicate which parts of the code he has written or feels otherwise responsible for.

You can use the following command to get list of files which you have touched in the past:

```shell
git log --name-status --pretty=format:'' --author="Steffen Vogel" | sort | uniq
```

I will start with my contributions:

```codeowners
/cmake/ @stv0g
/pyproject.toml @stv0g
/gitlab-ci.yml @stv0g
/dpsim/src/pybind @stv0g
/.clang-format @stv0g
/.editorconfig @stv0g
/.git-blame-ignore-revs @stv0g
/configs/shmem_cosim/ @stv0g

# Packaging
/.dockerignore @stv0g
/packaging/Docker @stv0g

# VILLAS-related
/dpsim-villas/ @stv0g
/examples/villas @stv0g
/examples/villas-deprecated @stv0g
```

n-eiling, 2024-04-24:
```
/dpsim/include/dpsim/GpuDenseAdapter.h
/dpsim/include/dpsim/GpuMagmaAdapter.h
/dpsim/include/dpsim/GpuSparseAdapter.h
/dpsim/include/dpsim/MNASolverPlugin.h
/dpsim/include/dpsim/MNASolverFactory.h
/dpsim/include/dpsim/MNASolverDynInterface.h
/dpsim/src/
/dpsim/src/
```

georgii-tishenin, 2024-04-24:
For me nothing in the master branch yet.

m-mirz, 2024-04-24:
WIP

```shell
/.dockerignore @m-mirz
/.github @m-mirz
/.editorconfig @m-mirz
/.gitignore @m-mirz
/.devcontainer @m-mirz
/CMakeLists.txt @m-mirz
/README.md @m-mirz
/cmake @m-mirz
/docs @m-mirz
/dpsim-models @m-mirz
/examples @m-mirz
/python @m-mirz
/requirements.txt @m-mirz
/setup.cfg @m-mirz
/sonar-project.properties @m-mirz
/configs @m-mirz
/dpsim @m-mirz
/packaging @m-mirz
#/pytest.ini @stv0g  ?
#/requirements-manylinux.txt @dinkelbachjan ?
/scripts @m-mirz
/setup.py @m-mirz
```

stv0g, 2024-05-02:
Quick reminder for: 

- @fwege
- @martinmoraga
- @JTS22
- @dinkelbachjan
- @gnakti
- @leonardocarreras

leonardocarreras, 2024-05-07:
From my side I could support with the workflows

`.github/workflows @leonardocarreras`

gnakti, 2024-05-07:
```codeowners
/dpsim-models @gnakti
/dpsim @gnakti
/examples @gnakti
/python @gnakti
```

leonardocarreras, 2024-05-07:
> WIP
>
> ```shell
> /.dockerignore @m-mirz
> /.github @m-mirz
> /.editorconfig @m-mirz
> /.gitignore @m-mirz
> /.devcontainer @m-mirz
> /CMakeLists.txt @m-mirz
> /README.md @m-mirz
> /cmake @m-mirz
> /docs @m-mirz [...]

Hi @stv0g! I think this one got missing, I do not see Markus owning anything in the PR... It says WIP, maybe @m-mirz wants to add something more?

stv0g, 2024-05-08:
@gnakti could you be a bit more specific one which files/models/directories you claim ownership?

The directories in your previous comment seem a bit too broad in my opinion..

stv0g, 2024-05-08:
Sorry @m-mirz, I forgot you add you in the PR. I revised the change..

m-mirz, 2024-05-09:
@stv0g no problem. Maybe it makes more sense to do it like in villas node? https://github.com/VILLASframework/node/blob/e26a5c1b48e4746839a83a8f942ac75cca25b3d2/CODEOWNERS#L8

This reduces the probability that a file is not covered by the CODEOWNERS file and I also like the idea that the more specific rules further down take precedence.

stv0g, 2024-05-13:
@m-mirz Yes I think that makes sense. I've added you and Jan to the catch-all entry.

martinmoraga, 2024-05-13:
> @gnakti could you be a bit more specific one which files/models/directories you claim ownership?
> 
> The directories in your previous comment seem a bit too broad in my opinion..

@m-mirz @gnakti 

I can take thee dpsim models that I implemented and once the branch sg-controllers take over the SG-controllers.  I could also take over the matpower reader if you guys agree.... 

 dpsim-models/include/dpsim-models/Base/Base_ReducedOrderSynchronGenerator.h                 @martinmoraga
dpsim-models/src/Base/Base_ReducedOrderSynchronGenerator.cpp                                              @martinmoraga                                                                                                                                                         

dpsim-models/include/dpsim-models/DP/DP_Ph1_SynchronGenerator3OrderVBR.h                      @martinmoraga 
dpsim-models/include/dpsim-models/EMT/EMT_Ph3_SynchronGenerator3OrderVBR.h                 @martinmoraga 
dpsim-models/include/dpsim-models/SP/SP_Ph1_SynchronGenerator3OrderVBR.h                       @martinmoraga 
dpsim-models/src/EMT/EMT_Ph3_SynchronGenerator3OrderVBR.cpp                                             @martinmoraga 
dpsim-models/src/SP/SP_Ph1_SynchronGenerator3OrderVBR.cpp                                                   @martinmoraga 
dpsim-models/src/DP/DP_Ph1_SynchronGenerator3OrderVBR.cpp                                                  @martinmoraga 

dpsim-models/include/dpsim-models/DP/DP_Ph1_SynchronGenerator4OrderVBR.h                      @martinmoraga 
dpsim-models/include/dpsim-models/EMT/EMT_Ph3_SynchronGenerator4OrderVBR.h                 @martinmoraga 
dpsim-models/include/dpsim-models/SP/SP_Ph1_SynchronGenerator4OrderVBR.h                       @martinmoraga 
dpsim-models/src/EMT/EMT_Ph3_SynchronGenerator4OrderVBR.cpp                                             @martinmoraga 
dpsim-models/src/SP/SP_Ph1_SynchronGenerator4OrderVBR.cpp                                                   @martinmoraga 
dpsim-models/src/DP/DP_Ph1_SynchronGenerator4OrderVBR.cpp                                                  @martinmoraga 

dpsim-models/include/dpsim-models/DP/DP_Ph1_SynchronGenerator5OrderVBR.h                       @martinmoraga 
dpsim-models/include/dpsim-models/EMT/EMT_Ph3_SynchronGenerator5OrderVBR.h                  @martinmoraga 
dpsim-models/include/dpsim-models/SP/SP_Ph1_SynchronGenerator5OrderVBR.h                        @martinmoraga 
dpsim-models/src/EMT/EMT_Ph3_SynchronGenerator5OrderVBR.cpp                                              @martinmoraga 
dpsim-models/src/SP/SP_Ph1_SynchronGenerator5OrderVBR.cpp                                                    @martinmoraga 
dpsim-models/src/DP/DP_Ph1_SynchronGenerator5OrderVBR.cpp                                                   @martinmoraga   

dpsim-models/include/dpsim-models/DP/DP_Ph1_SynchronGenerator6aOrderVBR.h                       @martinmoraga 
dpsim-models/include/dpsim-models/EMT/EMT_Ph3_SynchronGenerator6aOrderVBR.h                  @martinmoraga 
dpsim-models/include/dpsim-models/SP/SP_Ph1_SynchronGenerator6aOrderVBR.h                        @martinmoraga 
dpsim-models/src/EMT/EMT_Ph3_SynchronGenerator6aOrderVBR.cpp                                              @martinmoraga 
dpsim-models/src/SP/SP_Ph1_SynchronGenerator6aOrderVBR.cpp                                                    @martinmoraga 
dpsim-models/src/DP/DP_Ph1_SynchronGenerator6aOrderVBR.cpp                                                   @martinmoraga   

dpsim-models/include/dpsim-models/DP/DP_Ph1_SynchronGenerator6bOrderVBR.h                       @martinmoraga 
dpsim-models/include/dpsim-models/EMT/EMT_Ph3_SynchronGenerator6bOrderVBR.h                  @martinmoraga 
dpsim-models/include/dpsim-models/SP/SP_Ph1_SynchronGenerator6bOrderVBR.h                        @martinmoraga 
dpsim-models/src/EMT/EMT_Ph3_SynchronGenerator6bOrderVBR.cpp                                              @martinmoraga 
dpsim-models/src/SP/SP_Ph1_SynchronGenerator6bOrderVBR.cpp                                                    @martinmoraga 
dpsim-models/src/DP/DP_Ph1_SynchronGenerator6bOrderVBR.cpp                                                   @martinmoraga   

dpsim/examples/cxx/Circuits/DP_ReducedOrderSG_SMIB_Fault.cpp                                  
dpsim/examples/cxx/Circuits/DP_ReducedOrderSG_VBR_Load_Fault.cpp
dpsim/examples/cxx/Circuits/DP_SMIB_ReducedOrderSG_LoadStep.cpp
dpsim/examples/cxx/Circuits/EMT_ReducedOrderSG_SMIB_Fault.cpp
dpsim/examples/cxx/Circuits/EMT_ReducedOrderSG_VBR_Load_Fault.cpp
dpsim/examples/cxx/Circuits/EMT_SMIB_ReducedOrderSG_LoadStep.cpp
dpsim/examples/cxx/Circuits/SP_SMIB_ReducedOrderSG_LoadStep.cpp
dpsim/examples/cxx/Circuits/SP_ReducedOrderSG_SMIB_Fault.cpp
dpsim/examples/cxx/Circuits/SP_ReducedOrderSG_VBR_Load_Fault.cpp

  examples/Notebooks/Circuits/SP_Validation_ReducedOrderSG_VBR_SMIB_Fault_withPSAT.ipynb    @martinmoraga

n-eiling, 2024-06-06:
Can we make this happen? :)

gnakti, 2024-06-10:
> @gnakti could you be a bit more specific one which files/models/directories you claim ownership?
> 
> The directories in your previous comment seem a bit too broad in my opinion..

```codeowners
/dpsim-models/include/dpsim-models/CIM/Reader.h @gnakti
/dpsim-models/src/CIM/Reader.cpp @gnakti
/dpsim-models/include/dpsim-models/DP/DP_Ph1_SynchronGeneratorTrStab.h @gnakti
/dpsim-models/src/DP/DP_Ph1_SynchronGeneratorTrStab.cpp @gnakti
/dpsim-models/include/dpsim-models/SP/SP_Ph1_SynchronGeneratorTrStab.h @gnakti
/dpsim-models/src/SP/SP_Ph1_SynchronGeneratorTrStab.cpp @gnakti
/dpsim-models/include/dpsim-models/DP/DP_Ph1_varResSwitch.h @gnakti
/dpsim-models/src/DP/DP_Ph1_varResSwitch.cpp @gnakti
/dpsim-models/include/dpsim-models/SP/SP_Ph1_varResSwitch.h @gnakti
/dpsim-models/src/SP/SP_Ph1_varResSwitch.cpp @gnakti
/dpsim/include/dpsim/PFSolver.h @gnakti
/dpsim/src/PFSolver.cpp @gnakti
/dpsim/include/dpsim/PFSolverPowerPolar.h @gnakti
/dpsim/src/PFSolverPowerPolar.cpp @gnakti
/dpsim/examples/cxx/CIM @gnakti
/dpsim/examples/cxx/Circuits @gnakti
/python @gnakti
/examples/Notebooks/Circuits @gnakti
/examples/Notebooks//Grids @gnakti
```

stv0g, 2024-06-11:
> Can we make this happen? :)

I've added the latest additions by @gnakti and @martinmoraga.

Now awaiting further feedback and/or approval from somebody with merge permissions :)

pipeacosta, 2024-06-11:
I think I could support with this one:
```
# VILLAS-related
/examples/villas @pipeacosta
```

dinkelbachjan, 2024-06-15:
> @m-mirz Yes I think that makes sense. I've added you and Jan to the catch-all entry.

Thanks, that's fine for me! I would suggest to additionally include the explanatory comment lines like in the villas node file:
https://github.com/VILLASframework/node/blob/e26a5c1b48e4746839a83a8f942ac75cca25b3d2/CODEOWNERS#L4-L13
Like this it's clear how to interpret the file :)

stv0g, 2024-06-15:
>  I would suggest to additionally include the explanatory comment lines like in the villas node file

Nice idea :) I've added it to the PR.

@dinkelbachjan, @m-mirz Please feel free to merge once you are happy

closed 2024-06-20 by m-mirz