sogno-platform/dpsim #308: A typo in dpsim_classes_simulation.svg

georgii-tishenin, 2024-06-20:
`ODESolver` class is twice in the [dpsim_classes_simulation.svg](https://github.com/sogno-platform/dpsim/blob/master/docs/hugo/content/en/docs/Overview/Architecture/dpsim_classes_simulation.svg), used in [Architecture | DPsim](https://dpsim.fein-aachen.org/docs/overview/architecture/).

From the context it should be `ODESolver` and `DAESolver`. Shall I fix that?

stv0g, 2024-06-20:
That would be great :) Is there maybe another new solver class which we could add instead?

georgii-tishenin, 2024-06-20:
Good point. I was actually going to just replace one of `ODESolvers` by `DAESolver` in the figure. I assume this was a typo from the start.

Though inheritance diagram in [Solver class reference](https://sogno.energy/dpsim/doxygen/class_d_psim_1_1_solver.html) has more classes on the same level. The additional ones are `DiakopticsSolver`,   `ODEIntSolver` and `PFSolver`.

PR #310 (2024-06-27): Fix docu issues
This PR fixes two small documentation issues:

- Substitute duplicate `ODESolver` by `DAESolver` in UML diagram [dpsim_classes_simulation.svg](https://github.com/sogno-platform/dpsim/blob/master/docs/hugo/content/en/docs/Overview/Architecture/dpsim_classes_simulation.svg), used in [Architecture | DPsim](https://dpsim.fein-aachen.org/docs/overview/architecture/). Fixes #308 .
- Add missing links for literature references in [Dynamic Phasors | DPsim](https://sogno.energy/dpsim/docs/concepts/dyn-phasors/#bandpass-signals-and-baseband-representation). Fixes #219 .
commit 204eb276 (2024-06-27): Fix docu issues (#310)

This PR fixes two small documentation issues:

- Substitute duplicate `ODESolver` by `DAESolver` in UML diagram
[dpsim_classes_simulation.svg](https://github.com/sogno-platform/dpsim/blob/master/docs/hugo/content/en/docs/Overview/Architecture/dpsim_classes_simulation.svg),
used in [Architecture |
DPsim](https://dpsim.fein-aachen.org/docs/overview/architecture/). Fixes
#308 .
- Add missing links for literature references in [Dynamic Phasors |
DPsim](https://sogno.energy/dpsim/docs/concepts/dyn-phasors/#bandpass-signals-and-baseband-representation).
Fixes #219 .
closed 2024-06-27 by m-mirz