sogno-platform/dpsim #313: Getting Error fatal error: 'DPsim.h' file not found

morje-datta, 2024-07-03:
Hello,
Current I have install fedora 36 install on my desktop.I have tried to build the dpsim library by step from following page
https://dpsim.fein-aachen.org/docs/getting-started/build/

completed step mentioned on web pages. when i tried to run the WSCC-9bus_CIM or any other file getting error saying 'DPsim.h' file not found. Added the path for it but error remains the same. 

[image]

leonardocarreras, 2024-07-04:
Hi @morje-datta, welcome to DPSim! As far as our knowledge goes, Fedora 36 is fully supported and our CI is actually using that specific version to build and test.

We would be keen to help, but we need more information. If you are a beginner with DPSim, one good starting point is to try to compile using the docker image that already includes all the things you need to succeed. That being said, if you are trying to compile natively then you need to self-support yourself a bit, you may want to ensure you have all the required dependencies installed. You can figure that out from the docker container and/or the scripts. 

Another point to take in account, is that usually it is a little easier to compile using _gcc_ than _clang_, some library support is better. That does not seems to be your choice from the screenshot provided. In the documentation steps we are using `gcc`.

Additionally, two more things to try: 1) run `cmake` from the terminal as it is described in the documentation, and include the target (for this case `--target WSCC-9bus_CIM `) and 2) run the file generated in the `build` folder using the terminal for example `./build/dpsim/examples/cxx/WSCC-9bus_CIM` (and not via Visual Studio Code). Configuring `cmake` in VSC can be a tricky task.

If you need more support, feel free to provide more details here. Another option is to join the slack channel #sogno-dpsim (as described in the repo) and ask there.

morje-datta, 2024-07-04:
Hie @leonardocarreras above reply was help to me. Library was build using docker image. I have run the command(./build/dpsim/examples/cxx/WSCC-9bus_CIM) from terminal it got run and shown the output.

when i made changes to the WSCC-9bus_CIM.cpp file(change finalTime =0.1=>finalTime =1) and try to build it give me following error 

/usr/bin/ld: cannot find -lGSL_LIBRARY-NOTFOUND
/usr/bin/ld: cannot find -lGSL_CBLAS_LIBRARY-NOTFOUND
collect2: error: ld returned 1 exit status

[image]

morje-datta, 2024-07-11:
I have deleted the all file and started from scratch. It work for me.  Right now i am trying with python binding getting following error while running the program. 
I have build dpsimpy and dpsimpyvillas same as mentioned in documentation but getting error mentioned in screenshot

 
[image]

leonardocarreras, 2024-07-11:
Hi, I think it will surely have something to do with the way you setup your VScode. Remember you need to connect to the kernel inside the docker container, not into some other part.
So, after you build, you need to start the jupyter server in the container.
```shell
cd /dpsim/build
export PYTHONPATH=$(pwd):$(pwd)/../python/src/
cd /dpsim
jupyter lab --ip="0.0.0.0" --allow-root --no-browser
```

After this, you select Select another kernel -> Existing Jupyter Server
There you enter the url to the machine (address/port) and the password is the token you see in the log of the server in the docker container 
[image]

Finally, all should work
[image]

morje-datta, 2024-07-19:
Hello,

I have followed same and it working properly. Now i have run the notebook file matpower-cae9. can you guide me how we can use import/export values at runtime using VILLASnode.

[image]

leonardocarreras, 2024-11-12:
Discussion was continued here https://github.com/VILLASframework/node/issues/798, then closing on DPSim.