sogno-platform/dpsim #321: Building DPsim with InterfaceVillas does not seem to link libvillas and libjansson correctly

SystemsPurge, 2024-10-10:
Following this example: 
https://dpsim.fein-aachen.org/docs/tasks/create-simulation/
First of all, the cmakelists file should also have dpsim-villas next to dpsim in the target link libraries, otherwise it does not work.
Even then one gets a `error adding symbols: DSO missing from command line` twice. I manually fixed it by adding libjansson and libvillas-common into link.txt in the cmakefiles.
Most likely is that when cmake looks for the villas libraries to build dpsim-villas, it only finds libvillas, and not libvillas-common ( thats the first DSO error )
The second one is because libjansson is not linked correctly , and one gets: `undefined reference to symbol 'json_string_value'`
After adding both to link.txt , the example builds normally:
`[100%] Built target my-scenario`
TO REPRODUCE:
Run https://dpsim.fein-aachen.org/docs/tasks/create-simulation/ with a scenario that contains interfacevillas/interfaceworkervillas code.

pipeacosta, 2024-10-31:
I can help 😊

leonardocarreras, 2025-03-28:
Hi @pipeacosta @SystemsPurge, does #335 solve this? Can we close?

SystemsPurge, 2025-04-15:
@leonardocarreras Seems like the whole interfaces section covers enough of the issue yes. I second closing this

leonardocarreras, 2025-04-15:
Thanks, closing then!

closed 2025-04-15 by leonardocarreras