sogno-platform/dpsim #330: add guidelines for attributes

m-mirz, 2024-12-28:
- consistent rules when to make variables attributes
- reduce number of attributes
- parameters do not need to be attributes in most cases, only if they are set dynamically e.g. from villasweb

georgii-tishenin, 2025-03-29:
Agree, it would be good to have some DOs and DONTs for making variables attributes. May be also good to suggest alternatives for using attributes.

### One example:

The constant `mBaseVoltage` in [SP_Ph1_PiLine](https://github.com/sogno-platform/dpsim/blob/master/dpsim-models/include/dpsim-models/SP/SP_Ph1_PiLine.h) is made an Attribute:

`const Attribute<Real>::Ptr mBaseVoltage;`

The function `determineNodeBaseVoltages()` in [PFSolver](https://github.com/sogno-platform/dpsim/blob/master/dpsim/src/PFSolver.cpp) needs to get base voltage from each of the components. This is how it's now done for the line:

`baseVoltage_ = line->attributeTyped<CPS::Real>("base_Voltage")->get()`. 

It's also done same way for other components.

If there is no _other_ specific reason why `mBaseVoltage`  needs to be an Attribute, I would suggest e.g. to have a getter function, so that PFSolver can get the value of base voltage from components:
```

public:
    // Getter for mBaseVoltage
    Real getBaseVoltage() const {
        return mBaseVoltage;
    }
```

georgii-tishenin, 2025-03-29:
where would we add those guidelines? here: [Attributes](https://dpsim.fein-aachen.org/docs/overview/attributes/) ?