sogno-platform/dpsim #344: CI random errors when compiling with clang in Rocky

leonardocarreras, 2025-03-28:
**Describe the bug**
The pipeline fails for Rocky Linux compiling with clang, when is not the first time

**To Reproduce**
Steps to reproduce the behavior:
1. Make a PR
2. Push additional commits
3. Compilation starts to fail
4. Cleaning github actions cache solves
5. After deleting the cache, it pass again

**Expected behavior**
No problems when running the CI.

**Additional context**
It also has the same behaviour for the master branch, cache stops being useful randomly for this specifics.

stv0g, 2025-03-31:
I can confirm this. Its pretty annoying. But I dont have a clue yet why its exactly failing..

stv0g, 2025-04-03:
I've not seen any more failing Rocky / Clang builds after #360 was merged.

I propose to close this issue after some more consistently successful builds.

leonardocarreras, 2025-04-07:
I am closing this at the end of the week if there are no new reports. CI seems to be more stable now in this respect.

closed 2025-04-14 by stv0g