sogno-platform/dpsim #345: CI pipeline fails for PyPI

leonardocarreras, 2025-03-28:
**Describe the bug**
The PyPI package does not get updated.

**To Reproduce**
Steps to reproduce the behavior:
1. Merge to master
2. Error happens

```
 Checking dist/dpsim-1.1.1.tar.gz: ERROR    InvalidDistribution: Metadata is missing required fields: Name,        
         Version.                                                               
         Make sure the distribution includes the files where those fields are   
         specified, and is using a supported Metadata-Version: 1.0, 1.1, 1.2,   
         2.0, 2.1, 2.2, 2.3.   
```

**Expected behavior**
A normal update to the existing version of the package.

**Additional context**
See for example
https://github.com/sogno-platform/dpsim/actions/runs/14136412004/job/39609082806

leonardocarreras, 2025-04-01:
@stv0g there is a new error now, #353 did not work correctly

stv0g, 2025-04-01:
@leonardocarreras Okay, sorry. I've messed up the version of the action. This is now fixed by #354.

stv0g, 2025-04-01:
But, the sdist publish workflow still fails, due to an outdated version of the setuptools package which is currently included in the `sogno/dpsim:dev`:

https://github.com/sogno-platform/dpsim/actions/runs/14188658427/job/39748479103

There is a shortcoming, that this image appears not to be build by the CI itself :(
So, I have a pretty strong feeling, that I dont have provileges to do it manually, and also breaking a lot of other stuff while doing so :(

stv0g, 2025-04-01:
Btw. why was commit https://github.com/sogno-platform/dpsim/commit/a703987727c2fd77b1643e276c29d3fad4468550 authored by you? Wasn't this my own commit? Did you force-push?

stv0g, 2025-04-01:
I added another commit to the new PR which upgrade the setuptools version:
https://github.com/sogno-platform/dpsim/pull/354/commits/5103091407474e356260c2d276e426490aa1fe83

But this requires somebody to update the sogno/dpsim:dev image.

@m-mirz?

leonardocarreras, 2025-04-01:
> Btw. why was commit [a703987](https://github.com/sogno-platform/dpsim/commit/a703987727c2fd77b1643e276c29d3fad4468550) authored by you? Wasn't this my own commit? Did you force-push?

I only did a merge commit, I am not sure why, apparently is based on this https://github.com/sogno-platform/dpsim/commit/909b39b7fd21a9c5a48d50a6575c110e7febe43b

leonardocarreras, 2025-04-01:
> I added another commit to the new PR which upgrade the setuptools version: [5103091](https://github.com/sogno-platform/dpsim/commit/5103091407474e356260c2d276e426490aa1fe83)
> 
> But this requires somebody to update the sogno/dpsim:dev image.
> 
> [@m-mirz](https://github.com/m-mirz)?

we can try to do this from the actions
https://github.com/sogno-platform/dpsim/actions/workflows/container.yaml

but maybe if it is not pointing to the right villas version & co, it might happen to break some things

leonardocarreras, 2025-04-01:
More things from old libraries pop up
https://github.com/sogno-platform/dpsim/actions/runs/14193041202/job/39761878938#step:4:1624

leonardocarreras, 2025-04-01:
> I added another commit to the new PR which upgrade the setuptools version: [5103091](https://github.com/sogno-platform/dpsim/commit/5103091407474e356260c2d276e426490aa1fe83)
> 
> But this requires somebody to update the sogno/dpsim:dev image.
> 
> [@m-mirz](https://github.com/m-mirz)?

last setuptools makes problems with the `m2r` package, so there are more things to fix... for the moment the containers are rolled back to the previous existing working version

stv0g, 2025-04-03:
Fixed in https://github.com/sogno-platform/dpsim/commit/9ea4994659c57dc11967f46982829778eb7a3b17

First passing pipeline https://github.com/sogno-platform/dpsim/actions/runs/14239359599

stv0g, 2025-04-06:
Unfortunately still failing: https://github.com/sogno-platform/dpsim/actions/runs/14291634741/job/40053721342

@m-mirz This is due to a missing trusted publishing configuration for the [PyPi test instance](https://test.pypi.org/). I already added the trusted publisher to the main PyPi instance as I have the required permissions there. But I am lacking those on test.pypi.org.

Could you please also add me as an owner for the DPsim project there?

My profile is: https://test.pypi.org/user/stv0g/

stv0g, 2025-04-08:
Fixed :) https://github.com/sogno-platform/dpsim/actions/runs/14324163162/job/40154230000

Thanks @m-mirz for the permissions.

PR #353 (2025-04-01): fix(ci): Attempt to fix PyPi package publish
Attempt to fix #345
commit a7039877 (2025-04-01): fix(ci): Attempt to fix PyPi package publish (#353)

Attempt to fix #345
closed 2025-04-08 by stv0g